x-wrt/x-wrt #383: How to enable remote control for docker?

bachvnnvn, 2024-04-09:
Hi, 

I am using this build for x86: X-WRT-docker 24.01_b202403222222

I modified the file "/tmp/dockerd/daemon.json" and added: "hosts": ["unix:///var/run/docker.sock", "tcp://0.0.0.0:2375"]

So the file looks like this:
{
  "data-root": "/data/docker/",
  "log-level": "warn",
  "iptables": true,
  "hosts": ["unix:///var/run/docker.sock", "tcp://0.0.0.0:2375"]
}

But the docker is still not listen on the port 2375.

Is there any way to enable remote control for docker?

Thanks!

clawneo, 2024-04-09:
`访问设备的远程管理，请使用正确的地址格式: http://{mac}.x-wrt.dev/

例如:  http://112233445566.x-wrt.dev/`

https://x-wrt.dev/  用这玩意不行么

ptpt52, 2024-04-11:
`lsof -ni :2375`

bachvnnvn, 2024-04-11:
I tried that, but it returned nothing.
2375 is not listening.

bachvnnvn, 2024-04-11:
I found that, what ever I changed in the file "/tmp/dockerd/daemon.json", after restarting (or stop --> Edit the file --> Start docker) with this command "/etc/init.d/dockerd restart"
The file "/tmp/dockerd/daemon.json" will be reset to the default on ==> That is why it doesn't listen on the port 2375.

Could you please let me know how to keep the editted "/tmp/dockerd/daemon.json" file?

Thanks!

ptpt52, 2024-04-11:
try config: `/etc/config/dockerd`

bachvnnvn, 2024-04-11:
Not much information I can change in the file /etc/config/dockerd
I edited the file "/etc/init.d/dockerd" ==> commented out the below lines but it still didn't help.

	# reset configuration
	#rm -fr "${DOCKER_CONF_DIR}"
	#mkdir -p "${DOCKER_CONF_DIR}"