x-wrt/x-wrt #391: Xiaomi Mi Router 3G very slow speed over the wire to the router

K0kius, 2024-05-08:
Why can I have very low speed to the router via wire? For example, checking the speed using iperf3, my Wi-Fi speed to the router is higher.
[image]

ptpt52, 2024-05-08:
what speed via wire?

K0kius, 2024-05-08:
> what speed via wire?

Network bandwidth between the router and computer over a network cable. In theory, the throughput should be about 1 Gbit/s, but as you can see from the screenshot, it is only 100+ Mbit/s.

ptpt52, 2024-05-09:
try:
```
swconfig dev switch0 show  | grep link
```

and run `mpstat -P ALL 1` while test iperf3

K0kius, 2024-05-09:
root@X-WRT:~# swconfig dev switch0 show  | grep link
        link: port:0 link:down
        link: port:1 link:up speed:100baseT full-duplex
        link: port:2 link:up speed:1000baseT full-duplex
        link: port:3 link:down
        link: port:4 link:down
        link: port:5 link:up speed:1000baseT full-duplex
        link: port:6 link:up speed:1000baseT full-duplex
root@X-WRT:~# mpstat -P ALL 1
Linux 6.6.30 (X-WRT)    05/09/24        _mips_  (4 CPU)

02:26:10     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:11     all    0.25    0.00    2.01    0.00    0.00    2.76    0.00    0.00   94.99
02:26:11       0    1.01    0.00    0.00    0.00    0.00    0.00    0.00    0.00   98.99
02:26:11       1    0.00    0.00    3.96    0.00    0.00    0.00    0.00    0.00   96.04
02:26:11       2    0.00    0.00    2.02    0.00    0.00   11.11    0.00    0.00   86.87
02:26:11       3    0.00    0.00    2.00    0.00    0.00    0.00    0.00    0.00   98.00

02:26:11     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:12     all    0.26    0.00   10.20    0.00    0.00   21.68    0.00    0.00   67.86
02:26:12       0    1.03    0.00    9.28    0.00    0.00    0.00    0.00    0.00   89.69
02:26:12       1    0.00    0.00   10.20    0.00    0.00    0.00    0.00    0.00   89.80
02:26:12       2    0.00    0.00   15.00    0.00    0.00   85.00    0.00    0.00    0.00
02:26:12       3    0.00    0.00    6.12    0.00    0.00    0.00    0.00    0.00   93.88

02:26:12     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:13     all    1.05    0.00   28.53    0.00    0.00   26.44    0.00    0.00   43.98
02:26:13       0    1.02    0.00   41.84    0.00    0.00    1.02    0.00    0.00   56.12
02:26:13       1    2.17    0.00   41.30    0.00    0.00    0.00    0.00    0.00   56.52
02:26:13       2    0.00    0.00    0.00    0.00    0.00  100.00    0.00    0.00    0.00
02:26:13       3    1.10    0.00   32.97    0.00    0.00    0.00    0.00    0.00   65.93

02:26:13     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:14     all    1.64    0.00   20.77    0.00    0.00   27.32    0.00    0.00   50.27
02:26:14       0    3.49    0.00   41.86    0.00    0.00    0.00    0.00    0.00   54.65
02:26:14       1    3.30    0.00   40.66    0.00    0.00    0.00    0.00    0.00   56.04
02:26:14       2    0.00    0.00    0.00    0.00    0.00  100.00    0.00    0.00    0.00
02:26:14       3    0.00    0.00    3.33    0.00    0.00    0.00    0.00    0.00   96.67

02:26:14     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:15     all    1.08    0.00   26.83    0.00    0.00   27.37    0.00    0.00   44.72
02:26:15       0    1.25    0.00   43.75    0.00    0.00    1.25    0.00    0.00   53.75
02:26:15       1    3.19    0.00   61.70    0.00    0.00    0.00    0.00    0.00   35.11
02:26:15       2    0.00    0.00    0.00    0.00    0.00  100.00    0.00    0.00    0.00
02:26:15       3    0.00    0.00    6.32    0.00    0.00    0.00    0.00    0.00   93.68

02:26:15     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:16     all    2.39    0.00   31.03    0.00    0.00   26.79    0.00    0.00   39.79
02:26:16       0    5.32    0.00   43.62    0.00    0.00    0.00    0.00    0.00   51.06
02:26:16       1    2.27    0.00   52.27    0.00    0.00    1.14    0.00    0.00   44.32
02:26:16       2    0.00    0.00    0.00    0.00    0.00  100.00    0.00    0.00    0.00
02:26:16       3    2.11    0.00   31.58    0.00    0.00    0.00    0.00    0.00   66.32

02:26:16     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:17     all    0.54    0.00   27.61    0.00    0.00   26.81    0.00    0.00   45.04
02:26:17       0    1.15    0.00   48.28    0.00    0.00    0.00    0.00    0.00   50.57
02:26:17       1    1.14    0.00   56.82    0.00    0.00    0.00    0.00    0.00   42.05
02:26:17       2    0.00    0.00    0.00    0.00    0.00  100.00    0.00    0.00    0.00
02:26:17       3    0.00    0.00   11.22    0.00    0.00    0.00    0.00    0.00   88.78

02:26:17     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:18     all    1.93    0.00   25.90    0.00    0.00   27.55    0.00    0.00   44.63
02:26:18       0    3.53    0.00   45.88    0.00    0.00    0.00    0.00    0.00   50.59
02:26:18       1    3.75    0.00   62.50    0.00    0.00    0.00    0.00    0.00   33.75
02:26:18       2    0.00    0.00    0.00    0.00    0.00  100.00    0.00    0.00    0.00
02:26:18       3    1.03    0.00    5.15    0.00    0.00    0.00    0.00    0.00   93.81

02:26:18     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:19     all    1.93    0.00   34.71    0.00    0.00   27.55    0.00    0.00   35.81
02:26:19       0    1.18    0.00   65.88    0.00    0.00    0.00    0.00    0.00   32.94
[... 20 lines cut ...]
02:26:22       0    2.06    0.00    2.06    0.00    0.00    0.00    0.00    0.00   95.88
02:26:22       1    0.00    0.00    4.04    0.00    0.00    0.00    0.00    0.00   95.96
02:26:22       2    0.00    0.00    5.00    0.00    0.00    0.00    0.00    0.00   95.00
02:26:22       3    0.99    0.00    3.96    0.00    0.00    1.98    0.00    0.00   93.07

02:26:22     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
[... cut ...]
02:26:24     CPU    %usr   %nice    %sys %iowait    %irq   %soft  %steal  %guest   %idle
02:26:25     all    1.76    0.00    2.52    0.00    0.00    0.00    0.00    0.00   95.72
02:26:25       0    1.01    0.00    3.03    0.00    0.00    0.00    0.00    0.00   95.96
02:26:25       1    1.02    0.00    2.04    0.00    0.00    0.00    0.00    0.00   96.94
02:26:25       2    3.00    0.00    2.00    0.00    0.00    0.00    0.00    0.00   95.00
02:26:25       3    1.98    0.00    2.97    0.00    0.00    0.00    0.00    0.00   95.05

ptpt52, 2024-05-09:
port 1 is wan port for mi-3g, it is 100Mbps connected to uplink.

ptpt52, 2024-05-09:
you connect to lan port and test iperf3?

K0kius, 2024-05-09:
> port 1 is wan port for mi-3g, it is 100Mbps connected to uplink.

Perhaps this is due to the fact that I have an old cable from the provider and the Internet speed does not exceed 100 Mbit/s

> you connect to lan port and test iperf3?

Yes

ptpt52, 2024-05-09:
lan to router should be 600Mbps, your test is 125Mbps only?

K0kius, 2024-05-09:
Each time the speed is different but within +-100 - 200 Mbit/s
[image]

K0kius, 2024-05-09:
There is more bandwidth from a phone via 5 GHz wi-fi, although in theory there should be more through cable.

[image]

K0kius, 2024-05-09:
Network bandwidth on stock ope wrt much more.

[image]