x-wrt/x-wrt #392: Linksys EA4500 Web服务无法启动

excpu, 2024-05-09:
具体表现为刷入后路由器正常启动，WIFI可以搜索到，连接有线可以获取到HDCP但web管理无法进入
x-wrt-24.04-b202405071017-kirkwood-generic-linksys_ea4500-squashfs-factory

SHA1: 813a2844a5234c21f4471613b8e0df5208f66d9e

ptpt52, 2024-05-09:
新刷入的吗？还是升级上来的？

ascendbeing, 2024-05-09:
i don't speak your language but just a few hours ago i was talking about how if you're on 6.1 kernel openwrt, it'll always say you have to use firmware factory, and use fw_setenv commands every single time. with ea3500, same family as ea4500 and e4200v2 (kirkwood).
https://github.com/openwrt/openwrt/issues/14546#issuecomment-2101693114
related?
basically once you're on 6.1 with ea3500/ea4500, it will always prevent you from doing a normal sysupgrade. you will always have to fw_setenv and flash factory bin
and it probably doesn't have to be that way, open/xwrt ppl can probably figure this out somehow.

excpu, 2024-05-09:
> 新刷入的吗？还是升级上来的？
升级上来的

excpu, 2024-05-09:
> i don't speak your language but just a few hours ago i was talking about how if you're on 6.1 kernel openwrt, it'll always say you have to use firmware factory, and use fw_setenv commands every single time. with ea3500, same family as ea4500 and e4200v2 (kirkwood). [openwrt#14546 (comment)](https://github.com/openwrt/openwrt/issues/14546#issuecomment-2101693114) related? basically once you're on 6.1 with ea3500/ea4500/e4200v2, it will always prevent you from doing a normal sysupgrade. you will always have to fw_setenv and flash factory bin and it probably doesn't have to be that way, open/xwrt ppl can probably figure this out somehow.
> 
> EDIT: BTW im too lazy to translate your post via tool. but if you are having the issue I'm talking about, if your question is, is it safe to upgrade, i have done the fw_setenv and flash factory process like 2-4x. so yes it's safe to do. but it will always (at least last I checked) whinge that you have to do this every single time. i will say I've never used xwrt, just open wrt. i would def switch if xwrt fixed this.

I upgraded my device to xwrt then I found out the web service did not work on my device.

ptpt52, 2024-05-09:
After the boot is complete, press and hold the reset button for 6 seconds and then release it. This restores the default configuration and should solve the problem. This is a bug in a certain version.

excpu, 2024-05-09:
> After the boot is complete, press and hold the reset button for 6 seconds and then release it. This restores the default configuration and should solve the problem. This is a bug in a certain version.

I tried to restore the default configuration more than once, but it did not work.

ascendbeing, 2024-05-09:
> > After the boot is complete, press and hold the reset button for 6 seconds and then release it. This restores the default configuration and should solve the problem. This is a bug in a certain version.
> 
> I tried to restore the default configuration more than once, but it did not work.
are you "bricked"? can you ssh into it?
you shouldn't try to reset it the classic 30/30/30 way. that may cause problems.

on ea3500 and 4500, there are 2 partitions. to boot to the other partition, you have to unplug/turn off, while it's flashing the lights rapidly. lights go 1. rapid- > 2. slow->3. rapid.

so, 3 times in a row, unplug or power off (button) while it's on step 3 (2nd series of rapid flashing).

then plug it in (when you've done this 3x in a row, so 4th poweron) and you may be greeted with whatever's you may have had before, or with one of your flashes.

i dont personally recommend upgrading/installing higher than open wrt kernel 5.15. i think open wrt latest stable is 5.15, lastest snapshot is 6.1. the reason i don't recommend 6.1 is bc of the issue i mentioned. it may be related to the issue you're having. because it requires you do a bunch of manual stuff to install, i suspect it could require undocumented steps to flash onto a factory device.

If booting the alt partition works for you, i recomend installing open wrt 23.xx.x whatever the latest stable is, it'll get you to 5.15 or 5.10. relatively straightforward, no fw_setenv commands, all documentation will work on with that.

I'm saying open wrt just cause that's all i know, just use whatever version gives you 5.15 or older kernel if you can.

excpu, 2024-05-10:
> on ea3500 and 4500, there are 2 partitions. to boot to the other partition, you have to unplug/turn off, while it's flashing the lights rapidly. lights go 1. rapid- > 2. slow->3. rapid.
>
> so, 3 times [...]

I dont think it is bricked since the wireless and hdcp did work but both ssh and web interface are not usable. however i will try the 2nd partition way.

excpu, 2024-05-10:
I get back to the kernel version 5.4.255 and everything is working properly and I believe I will stick to this version. Thanks for your help.

closed 2024-05-10 by excpu

ptpt52, 2024-05-10:
@excpu 
could you help test what happens to latest version?
1. change config in web ui, enable ssh login
2. sysupgrade to lastest version then see what happens to the web server
3. if web access is down, then login ssh to see what is going on?
```
lsof -ni :80
/etc/init.d/nginx restart
logread -e nginx
```