x-wrt/x-wrt #393: Xiaomi Router 3 编译失败 ramips mt7620 xiaomi_miwifi-r3

luojixinhao, 2024-05-31:
。。。。。

 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/x-wrt/include/toplevel.mk:233: world] Error 1
make[2]: Entering directory '/workdir/x-wrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/x-wrt/scripts/config'
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/x-wrt/include/toplevel.mk:233: world] Error 1

。。。。。。。

patching file net/dsa/slave.c
Patch failed!  Please fix /workdir/x-wrt/target/linux/ramips/patches-6.6/995-0001-hwnat-add-natflow-flow-offload-support.patch!
make[4]: *** [Makefile:32: /workdir/x-wrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-6.6.32/.prepared_5eb353e245e9d87ec3efb8d8aec645d6] Error 1
make[4]: Leaving directory '/workdir/x-wrt/target/linux/ramips'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/workdir/x-wrt/target/linux'
time: target/linux/compile#8.98#8.44#12.71
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/x-wrt'
make[1]: *** [target/Makefile:25: /workdir/x-wrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/x-wrt'
make: *** [/workdir/x-wrt/include/toplevel.mk:233: world] Error 2
Error: Process completed with exit code 2.

ptpt52, 2024-05-31:
编译教程 https://blog.x-wrt.com/docs/build/
选择最新tag版本编译

luojixinhao, 2024-05-31:
> 编译教程 https://blog.x-wrt.com/docs/build/ 选择最新tag版本编译

谢谢，搞定