x-wrt/x-wrt #394: Xiaomi Mi Route 3 从当前安装的LEDE R24.5.1更换到X-Wrt需要如何操作？

luojixinhao, 2024-05-31:
直接从备份升级中选择刷写固件“x-wrt-24.04-b202405311437-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin” 等候一段时间依然是原系统。
用命令sysupgrade -v x-wrt-24.04-b202405302325-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin 也一样刷不进去呢？

ptpt52, 2024-06-01:
刷了第三方uboot了吗？

luojixinhao, 2024-06-01:
> 刷了第三方uboot了吗？

没刷，之前刷了个pb-boot, 后面不知道怎么没了

ptpt52, 2024-06-01:
dmesg
cat /proc/mtd
执行这两个命令，发出来看看。

luojixinhao, 2024-06-01:
> dmesg cat /proc/mtd 执行这两个命令，发出来看看。

我已经先刷回官方，再刷X-WRT了

luojixinhao, 2024-06-01:
> dmesg cat /proc/mtd 执行这两个命令，发出来看看。

这是另一个路由。装的是官方openwrt。可以直接刷成X-wrt吗？

root@mi_r3p:~# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00020000 "Bootloader"
mtd1: 00040000 00020000 "Config"
mtd2: 00040000 00020000 "Bdata"
mtd3: 00040000 00020000 "factory"
mtd4: 00040000 00020000 "crash"
mtd5: 00080000 00020000 "crash_syslog"
mtd6: 00040000 00020000 "reserved0"
mtd7: 00400000 00020000 "kernel_stock"
mtd8: 00400000 00020000 "kernel"
mtd9: 0f580000 00020000 "ubi"

root@mi_r3p:~# dmesg
[    0.000000] Linux version 6.6.30 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 13.2.0 r26435-f7f8099aa3) 13.2.0, GNU ld (GNU Binutils) 2.42) #0 SMP Sun May 26 22:32:57 2024
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Xiaomi Mi Router 3 Pro
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000001fffffff]
。。。。

ptpt52, 2024-06-01:
可以的。直接sysupgrade升级

luojixinhao, 2024-06-01:
> 可以的。直接sysupgrade升级

好的，我稍后试试