x-wrt/x-wrt #402: How to flash x-wrt on JCG Q20?

bachvnnvn, 2024-06-28:
Hi,

I have a JCG Q20 access point.

I downloaded the openwrt firmware from this link: https://firmware-selector.openwrt.org/?version=23.05.3&target=ramips%2Fmt7621&id=jcg_q20

1. I flashed the [KERNEL](https://downloads.openwrt.org/releases/23.05.3/targets/ramips/mt7621/openwrt-23.05.3-ramips-mt7621-jcg_q20-initramfs-kernel.bin) ==> Worked with blue light

2. I flashed x-wrt: https://downloads.x-wrt.com/rom/x-wrt-24.04-b202406210000-ramips-mt7621-jcg_q20-squashfs-sysupgrade.bin ==> DIDN'T WORK with a red light. And it went back to the recovery mode (PandoraBox).

Could you please tell me where I was wrong?

ptpt52, 2024-06-28:
try factory.bin?

bachvnnvn, 2024-06-29:
Where can I get that file?

bachvnnvn, 2024-06-29:
Yes, already tried that file as well.

The same issue.

ptpt52, 2024-06-29:
you should intall it from stock firmware's upgrade page

closed 2024-07-02 by bachvnnvn