x-wrt/x-wrt #409: Xiaomi Mi Wifi Router 3 - 28/08/2024 fresh install "x-wrt-24.04-b202408260930-ramips-mt7620"

jpvfoliveira, 2024-08-28:
Hello, I just installed the latest version of X-WRT on my router but it keeps rebooting from time to time. Can someone please help?

ptpt52, 2024-08-29:
I notice this bug, Please use old version for now.

ptpt52, 2024-08-29:
try test this firmware: http://downloads.x-wrt.com/rom/Downloads/x-wrt-24.04-b202408271653-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin

ptpt52, 2024-08-29:
try this one again:
https://downloads.x-wrt.com/rom/Downloads/x-wrt-24.04-b202408291429-ramips-mt7620-xiaomi_miwifi-r3-squashfs-sysupgrade.bin

ptpt52, 2024-11-01:
likely stable for now.

Bob-YsPan, 2024-11-04:
I set the router to reboot at every Monday's midnight; try to test it will reboot or freeze or not after 1 week.

If it does not reboot within this week, it is maybe stable enough, I think.

But maybe we still need some volunteers to try to use the new firmware to check it stable or not.

Bob-YsPan, 2024-11-10:
@jpvfoliveira I tried this version that release just few weeks ago: 
[https://github.com/x-wrt/build-release/releases/tag/v24.04_b202410201421](https://github.com/x-wrt/build-release/releases/tag/v24.04_b202410201421)

And it can run stable at least one week
Maybe you can try that

[image]

Bob-YsPan, 2025-11-19:
> What Version now you can advise?

Latest version.
It includes fixes for the freezing issue and the latest patches.
You can give it a try.