x-wrt/x-wrt #415: NanoPi R5C there are no drivers for MT7925

IIIef, 2024-10-04:
Good afternoon, could you add the drivers for the MT7925 card
[image]
[image]

ptpt52, 2024-10-05:
pcie card or usb card?

IIIef, 2024-10-05:
> pcie card or usb card?

pcie 
[image]

IIIef, 2024-10-13:
@ptpt52 in the last build, I look at the driver added, but it is not installed. please fix it
[image]
[image]
[image]
[image]