x-wrt/x-wrt #422: 鲲鹏c8  668

qgzqgz, 2024-11-07:
能把鲲鹏c8  668 dts和mt7981.mk加上吗？第二系统dts,668可以同时有两个系统，官方和op,我这里有第二系统dts和mk文件,如果行就把发上来，文件是5.4的，版本太老了

ptpt52, 2024-11-07:
和 C8-660 什么区别？

qgzqgz, 2024-11-09:
就是闪存和内存不一样668是1加8的

qgzqgz, 2024-11-09:
就是闪存和内存不一样668是1加8的

qgzqgz, 2024-11-09:
[新建 文本文档 (2).zip](https://github.com/user-attachments/files/17686931/2.zip)
[新建文件夹.zip](https://github.com/user-attachments/files/17686938/default.zip)

qgzqgz, 2024-11-09:
[mt7981.mk .zip](https://github.com/user-attachments/files/17687001/mt7981.mk.zip)

qgzqgz, 2024-11-09:
[mt7981.dts.zip](https://github.com/user-attachments/files/17687010/mt7981.dts.zip)

qgzqgz, 2024-11-09:
这是5.4分支版本里的文件

qgzqgz, 2024-11-09:
如果把这些文件添加到你的源码里编译就会报错，可能是版本的问题吧

ptpt52, 2024-11-29:
https://github.com/x-wrt/x-wrt/commit/626c854c6142b42f8c6f86dc836cbdccdaeb6716

ptpt52, 2024-11-29:
https://downloads.x-wrt.com/rom/Downloads/x-wrt-24.04-b202411262136-mediatek-filogic-nradio_c8-668gl-squashfs-sysupgrade.bin

qgz1, 2024-11-29:
> https://downloads.x-wrt.com/rom/Downloads/x-wrt-24.04-b202411262136-mediatek-filogic-nradio_c8-668gl-squashfs-sysupgrade.bin

这是编译好的固件下载链接吗？

ptpt52, 2024-11-29:
你刷了试试

qgz1, 2024-11-29:
a
***@***.***

 
刷好了可以用