x-wrt/x-wrt #426: Fake mesh Controller bridge mode doesn't work

bachvnnvn, 2024-11-11:
Hi,

I am using Mikrotik OS as the main router.
I have 4 AP and they are running x-wrt with fake mesh.
On 1 AP, I set it ad Controller and set statics IP for the Lan port.
On 3 AP, I set it as Wired AP agent.
But on 3 AP, it cannot detect the Controller.

Could you please guide me how to fix it?

Thanks!
[image]

ptpt52, 2024-11-11:
get the test info:

Controller: what the lan ip is?

on AP: `ubus call umdns browse`

bachvnnvn, 2024-11-12:
@ptpt52

Hi,

The Controller's IP is: 192.168.100.252

Below is the result of ubus that I did on 1 of the agent AP (192.168.100.250)

`root@MESH-AGENT_5033F0AC80E4:~# ubus call umdns browse
{
        "_dawn._tcp": {
                "MESH-AGENT_5033F0578B1C": {
                        "iface": "br-lan",
                        "host": "MESH-AGENT_5033F0578B1C.local",
                        "port": 1026,
                        "ipv4": "192.168.100.251",
                        "ipv6": "fe80::5233:f0ff:fe57:8b1c"
                },
                "T4": {
                        "iface": "br-lan",
                        "host": "T4.local",
                        "port": 1026,
                        "ipv4": "192.168.100.252",
                        "ipv6": "fe80::5233:f0ff:fe3c:1c90"
                }
        },
        "_fakemesh_PCRouter._tcp": {
                "T4": {
                        "iface": "br-lan",
                        "host": "T4.local",
                        "port": 65535,
                        "txt": "id=PCRouter",
                        "ipv4": "192.168.100.252",
                        "ipv6": "fe80::5233:f0ff:fe3c:1c90"
                }
        },
        "_googlecast._tcp": {
                "Google-Nest-Hub-xxxxxxxxxxxxxxxxxxxx": {
                        "iface": "br-lan",
                        "txt": "id=xxxxxxxxxxxxxxxxxxxxxxxxxxx",
                        "txt": "cd=xxxxxxxxxxxxxxxxxxxxxxxx",
                        "txt": "rm=xxxxxxxxxxxxxxxxxxxxxxxxx",
                        "txt": "ve=05",
                        "txt": "md=Google Nest Hub",
                        "txt": "ic=/setup/icon.png",
                        "txt": "fn=Bedroom display",
                        "txt": "ca=xxxxxxxxxxxxxxxxxxx",
                        "txt": "st=0",
                        "txt": "bs=xxxxxxxxxxxxxxxxxxxx",
                        "txt": "nf=1",
                        "txt": "rs=",
                        "host": "fuchsia-xxxxxxxxxxxxxxxxxx",
                        "port": xxxxxxxxxxxxxxxxxxxxxxxxx,
                        "ipv4": "192.168.100.102",
                        "ipv6": "fd07:97ef:93f2:5780:3c33:bc87:4f26:6105",
                        "ipv6": "fd07:97ef:93f2:5780:5279:a460:3bbd:7881",
                        "ipv6": "fe80::9704:46b:d745:7b55"
                },
                "Google-Nest-Hub-xxxxxxxxxxxxxxxxxxxxx": {
                        "iface": "br-lan",
                        "txt": "id=xxxxxxxxxxxxxxxxxxxxxxxx",
                        "txt": "cd=xxxxxxxxxxxxxxxxxxxxxxxxxx",
                        "txt": "rm=xxxxxxxxxxxxxxxxxxxxxx",
                        "txt": "ve=05",
                        "txt": "md=Google Nest Hub",
                        "txt": "ic=/setup/icon.png",
                        "txt": "fn=Living Room display",
                        "txt": "ca=xxxxxxxxxxxxxxx",
                        "txt": "st=0",
                        "txt": "bs=xxxxxxxxxxxxxxxxxxxxxxx",
[... 88 lines cut ...]
                        "host": "MESH-AGENT_5033F0578B1C.local",
                        "port": 22,
                        "txt": "daemon=dropbear",
                        "ipv4": "192.168.100.251",
                        "ipv6": "fe80::5233:f0ff:fe57:8b1c"
                },
                "T4": {
                        "iface": "br-lan",
                        "host": "T4.local",
                        "port": 22,
                        "txt": "daemon=dropbear",
                        "ipv4": "192.168.100.252",
                        "ipv6": "fe80::5233:f0ff:fe3c:1c90"
                }
        },
        "_teamviewer._tcp": {
                "xxxxxxxxxxxxxx": {
                        "iface": "br-lan",
                        "txt": "DyngateID=xxxxxxxxxxxxx",
                        "txt": "Token=xxxxxxxxxxxxxxxxx",
                        "txt": "UUID=xxxxxxxxxxxxxxxxxxxxxxxxxxx"
                }
        }
}
`

ptpt52, 2024-11-12:
AP 已经拿到 ac ip了

ptpt52, 2024-11-12:
确保 meshid都填写正确，PCRouter

bachvnnvn, 2024-11-12:
@ptpt52 yes, all the APs have the same mesh ID "PCRouter".
But as you can see in the snapshot above, on the Agent, the list of wireless is not there while it is on the Controller.

How to debug this further?

Thank you!

bachvnnvn, 2024-11-12:
Here is the setting on the Controller

[image]

bachvnnvn, 2024-11-12:
Here is on the Agent

[image]

ptpt52, 2024-11-12:
logread -e fakemesh

bachvnnvn, 2024-11-12:
> logread -e fakemesh

The log shows that the Agent is trying to get mesh info from the main router (192.168.100.1) and it is running Mikrotik OS.

`root@MESH-AGENT_5033F0578B1C:~# logread -e fakemesh
Tue Nov 12 10:39:09 2024 user.notice fakemesh: agent_sync_config(6533): started!
Tue Nov 12 10:46:58 2024 user.notice fakemesh: agent_sync_config(6533): umdns: achost not found(lost_ac=0)!
Tue Nov 12 10:46:58 2024 user.notice fakemesh: agent_sync_config(6533): fetch config from controller.fakemesh failed(1)!
Tue Nov 12 10:46:58 2024 user.notice fakemesh: agent_sync_config(6533): fetch config from achost(controller.fakemesh)/gateway(192.168.100.1) failed(1), wait 30s
Tue Nov 12 10:47:49 2024 user.notice fakemesh: agent_sync_config(6533): umdns: achost not found(lost_ac=1)!
Tue Nov 12 10:47:49 2024 user.notice fakemesh: agent_sync_config(6533): fetch config from controller.fakemesh failed(2)!
Tue Nov 12 10:47:49 2024 user.notice fakemesh: agent_sync_config(6533): fetch config from achost(controller.fakemesh)/gateway(192.168.100.1) failed(2), wait 30s
Tue Nov 12 10:48:39 2024 user.notice fakemesh: agent_sync_config(6533): umdns: achost not found(lost_ac=2)!
Tue Nov 12 10:48:39 2024 user.notice fakemesh: agent_sync_config(6533): fetch config from controller.fakemesh failed(3)!
Tue Nov 12 10:48:39 2024 user.notice fakemesh: agent_lost_ac(9722): started! lost_ac=3
Tue Nov 12 10:48:39 2024 user.notice fakemesh: agent_lost_ac(9722): setup temporary lan ip=192.168.16.1
Tue Nov 12 10:48:40 2024 user.notice fakemesh: agent_lost_ac(9722): starting wifi SSID=\'X-WRT_1C90\' on radio1 5g!
Tue Nov 12 10:48:49 2024 user.notice fakemesh: agent_lost_ac(9722): done!
Tue Nov 12 10:48:49 2024 user.notice fakemesh: agent_sync_config(11565): started!
Tue Nov 12 10:49:11 2024 user.notice fakemesh: agent_sync_config(11565): umdns: achost not found(lost_ac=3)!
Tue Nov 12 10:49:11 2024 user.notice fakemesh: agent_sync_config(11565): fetch config from controller.fakemesh failed(4)!
Tue Nov 12 10:49:11 2024 user.notice fakemesh: agent_lost_ac(14359): started! lost_ac=4
Tue Nov 12 10:49:11 2024 user.notice fakemesh: agent_sync_config(14392): started!
Tue Nov 12 10:49:34 2024 user.notice fakemesh: agent_sync_config(14392): umdns: achost not found(lost_ac=4)!
Tue Nov 12 10:49:34 2024 user.notice fakemesh: agent_sync_config(14392): fetch config from controller.fakemesh failed(5)!
Tue Nov 12 10:49:34 2024 user.notice fakemesh: agent_lost_ac(17290): started! lost_ac=5
Tue Nov 12 10:49:34 2024 user.notice fakemesh: agent_sync_config(17295): started!
Tue Nov 12 10:49:56 2024 user.notice fakemesh: agent_sync_config(17295): umdns: achost not found(lost_ac=5)!
Tue Nov 12 10:49:56 2024 user.notice fakemesh: agent_sync_config(17295): fetch config from controller.fakemesh failed(6)!
Tue Nov 12 10:49:56 2024 user.notice fakemesh: agent_lost_ac(20226): started! lost_ac=6
Tue Nov 12 10:49:56 2024 user.notice fakemesh: agent_sync_config(20231): started!
Tue Nov 12 10:50:19 2024 user.notice fakemesh: agent_sync_config(20231): umdns: achost not found(lost_ac=6)!
Tue Nov 12 10:50:19 2024 user.notice fakemesh: agent_sync_config(20231): fetch config from controller.fakemesh failed(7)!
Tue Nov 12 10:50:19 2024 user.notice fakemesh: agent_lost_ac(23170): started! lost_ac=7
Tue Nov 12 10:50:19 2024 user.notice fakemesh: agent_sync_config(23175): started!
Tue Nov 12 10:50:41 2024 user.notice fakemesh: agent_sync_config(23175): umdns: achost not found(lost_ac=7)!
Tue Nov 12 10:50:41 2024 user.notice fakemesh: agent_sync_config(23175): fetch config from controller.fakemesh failed(8)!
Tue Nov 12 10:50:41 2024 user.notice fakemesh: agent_lost_ac(26089): started! lost_ac=8
Tue Nov 12 10:50:41 2024 user.notice fakemesh: agent_sync_config(26094): started!
Tue Nov 12 10:51:04 2024 user.notice fakemesh: agent_sync_config(26094): umdns: achost not found(lost_ac=8)!
Tue Nov 12 10:51:04 2024 user.notice fakemesh: agent_sync_config(26094): fetch config from controller.fakemesh failed(9)!
Tue Nov 12 10:51:04 2024 user.notice fakemesh: agent_lost_ac(28938): started! lost_ac=9
Tue Nov 12 10:51:04 2024 user.notice fakemesh: agent_sync_config(28943): started!
Tue Nov 12 10:51:26 2024 user.notice fakemesh: agent_sync_config(28943): umdns: achost not found(lost_ac=9)!
Tue Nov 12 10:51:26 2024 user.notice fakemesh: agent_sync_config(28943): fetch config from controller.fakemesh failed(10)!
Tue Nov 12 10:51:26 2024 user.notice fakemesh: agent_lost_ac(31846): started! lost_ac=10
Tue Nov 12 10:51:26 2024 user.notice fakemesh: agent_sync_config(31851): started!
`

ptpt52, 2024-11-12:
AP上执行执行获取AC的ip，是否能获取
```
ubus call umdns browse | jsonfilter -e "@['_fakemesh_ PCRouter._tcp'][*]['ipv4']"
```

ptpt52, 2024-11-12:
多了个空格，更新一下命令

ptpt52, 2024-11-12:
```
ubus call umdns browse | jsonfilter -e "@['_fakemesh_PCRouter._tcp'][*]['ipv4']"
```

bachvnnvn, 2024-11-12:
@ptpt52 
+ I just restarted the AP (192.168.100.250) and the command "ubus call umdns browse" doesn't return the IP of AC (192.168.100.252) anymore.
+ 192.168.100.250 can ping 192.168.100.252
+ "ubus call umdns browse | jsonfilter -e "@['_fakemesh_PCRouter._tcp'][*]['ipv4']"" returns **nothing**

ptpt52, 2024-11-12:
re-run `ubus call umdns browse` multi-times after reboot.

bachvnnvn, 2024-11-12:
> re-run `ubus call umdns browse` multi-times after reboot.

I ran "ubus call umdns browse" more than 20 times then restarted the AP. Ran the "ubus call umdns browse" again but still no AC's IP there (192.168.100.252)

ptpt52, 2024-11-12:
为什么你之前的运行可以获得 ac IP 呢

bachvnnvn, 2024-11-12:
我也不知道。不过在重启AP之后，Controller的IP不再出现了，尽管我仍然可以ping通。有没有办法手动调整让AP接收Controller的IP呢？

bachvnnvn, 2024-11-12:
挺有趣的。我进入AP并取消勾选Fake Mesh，AC的IP出现了。如果我勾选Fake Mesh，AC的IP就消失了。

bachvnnvn, 2024-11-12:
I ran this command on each AP: echo "192.168.100.252 controller.fakemesh" >> /etc/hosts

And now the new log:

```
root@MESH-AGENT_5033F0AC80E4:~# echo "192.168.100.252 controller.fakemesh" >> /etc/hosts
root@MESH-AGENT_5033F0AC80E4:~# ubus call umdns browse | jsonfilter -e "@['_fakemesh_PCRouter._tcp'][*]['ipv4']"
192.168.100.252
root@MESH-AGENT_5033F0AC80E4:~# logread -e fakemesh
Tue Nov 12 13:28:58 2024 daemon.err uhttpd[2184]: [info] luci: accepted login on /admin/mesh/fakemesh for admin from 192.168.100.9
Tue Nov 12 13:33:12 2024 user.notice fakemesh: agent_sync_config(29587): started!
Tue Nov 12 13:33:28 2024 user.notice fakemesh: agent_sync_config(29587): fetch config from 192.168.100.252 failed(1)!
Tue Nov 12 13:33:28 2024 user.notice fakemesh: agent_sync_config(29587): fetch config from achost(192.168.100.252)/gateway(192.168.100.1) failed(1), wait 30s

```