x-wrt/x-wrt #427: Can not run opkg update

thientruongbka, 2024-11-14:
Hi, 
I see all links in /etc/opkg/distfeeds.conf are not available, actually return 404 status
So I can't run opkg update
Please provide the alternative link to get package.gz files to run opkg update

[image]

[image]

Thanks

ptpt52, 2024-11-14:
https://github.com/x-wrt/build-release/issues/55

thientruongbka, 2024-11-15:
@ptpt52 so wait for next release right?

congphat1998, 2024-11-16:
/etc/opkg/distfeeds.conf

src/gz x-wrt_core http://downloads.openwrt.org/releases/packages-24.10/x86_64/packages
src/gz x-wrt_base http://downloads.openwrt.org/releases/packages-24.10/x86_64/base
src/gz x-wrt_luci http://downloads.openwrt.org/releases/packages-24.10/x86_64/luci
src/gz x-wrt_packages http://downloads.openwrt.org/releases/packages-24.10/x86_64/packages
src/gz x-wrt_routing http://downloads.openwrt.org/releases/packages-24.10/x86_64/routing
src/gz x-wrt_telephony http://downloads.openwrt.org/releases/packages-24.10/x86_64/telephony

Change x86_64 to your device

Bob-YsPan, 2024-12-01:
I tried modified @congphat1998 's source list for my MT7620 device, but can't passed the signature check

Source lists I modified:
```
src/gz x-wrt_core http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/packages
src/gz x-wrt_base http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/base
src/gz x-wrt_luci http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/luci
src/gz x-wrt_packages http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/packages
src/gz x-wrt_routing http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/routing
src/gz x-wrt_telephony http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/telephony
```

The errors like this:
```
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_core
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_base
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/base/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_luci
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/luci/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_packages
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_routing
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/routing/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/x-wrt_telephony
Downloading http://downloads.openwrt.org/releases/packages-24.10/mipsel_24kc/telephony/Packages.sig
Signature check failed.
Remove wrong Signature file.
```

Bob-YsPan, 2024-12-01:
> I tried modified @congphat1998 's source list for my MT7620 device, but can't passed the signature check

I switched to version 23.05 of the repository and was able to update the list, but I feel like I might run into some package conflicts, but I've installed a few of them so far and they all work fine.

menesc, 2024-12-01:
You should manually install some packages fist to prevent ssl errors. I don't remember exactly but i think 
ca-bundle_20240203-r1_all.ipk and libopenssl3_3.0.15-r1_mipsel_24kc.ipk

Bob-YsPan, 2024-12-02:
I upgraded ca-bundle and install libopenssl3 from 24.10 mirror, but still can't check the signature.

But I use the OpenWRT official's opkg verbose script, it can update software list, and even no need the libopenssl3 package:
[image]

Maybe some command failed so the opkg update can't check the signature, so still need to point out what the problem is.

ptpt52, 2024-12-02:
ok, should ship with the Public usign key for 24.10 release builds.

menesc, 2024-12-08:
I confirm the problem solved with "X-WRT v24.04_b202412081117" release on ramips-mt7620 (xiaomi_miwifi-r3).