x-wrt/x-wrt #428: pptpd: unrecognized option 'mppe'

Yish1, 2024-11-21:
作者您好，在使用24.11.20的xwrt固件时，遇到pptp存在问题，设备日志中看到Thu Nov 21 16:07:01 2024 daemon.err pppd[20613]: In file /var/etc/options.pptpd: unrecognized option 'mppe'    导致客户端无法连接，在ssh中修改后正常。
另外想问一下固件是否支持mu-mimo？ 2401Mbps局域网互传文件速度只有90mb/s多。

Hello, author. When using the xwrt firmware version 24.11.20, I encountered an issue with PPTP. The device logs show the following error:
Thu Nov 21 16:07:01 2024 daemon.err pppd[20613]: In file /var/etc/options.pptpd: unrecognized option 'mppe'
This caused the client to fail to connect. After modifying it via SSH, the connection worked fine.

Additionally, I would like to ask if the firmware supports MU-MIMO? The file transfer speed in the LAN only reaches around 90 MB/s, despite a 2401 Mbps connection.
Thanks!

Yish1, 2024-11-21:
device: Xiaomi WR30U

ptpt52, 2024-11-23:
pptpd的问题是因为老系统升级固件上来导致的吗？恢复默认配置后正常吗

mu-mimo 默认支持的

Yish1, 2024-11-23:
不是老系统升级，直接刷的，小米r3g和wr30u都有遇到相同问题

ptpt52, 2024-11-23:
一键vpn 开启pptpd不存在你说的问题。你如何开启pptpd呢，如何配置呢

Yish1, 2024-11-23:
[image]
就这样开的

Yish1, 2024-11-23:
手动配置的l2tp就能连上，pptp死活连不上ε(┬┬﹏┬┬)3

ptpt52, 2024-11-23:
局域网连你能连上吗，连不上很正常，现在pptp很少用了，不知道客户端那边支持情况

Yish1, 2024-11-23:
是在局域网里测试的，
[image]
不过也没什么了，l2tp就可以🤣😂

closed 2024-11-23 by Yish1