x-wrt/x-wrt #432: 你上传c8 668lg机型编译出来的固件只能刷单系统，现在在用的这款cpe都是用的双系统，在双系统下就不能用了

qgz1, 2024-11-29:


ptpt52, 2024-11-29:
能用呀，刷到第二个系统去

ptpt52, 2024-11-29:
你刷了主要看个方面功能是否正常，mac地址是否ok

qgz1, 2024-11-29:
a
***@***.***

 
用你源码编译过660刷到第二系统试过，就是反复重启

ptpt52, 2024-11-29:
668固件上传了，能刷吗，能跑吗

qgz1, 2024-11-29:
a
***@***.***

 
我现在用的668不能刷单系统了

ptpt52, 2024-11-29:
刷不进二系统吗？

ptpt52, 2024-11-29:
你登录一下原始系统，执行一下 cat /tmp/sysinfo/board_name
看看输出是什么

qgz1, 2024-11-29:
a
***@***.***

 
刷是能刷进去可能会反复重启，进不了系统，因为专门做这个型号人说过单系统和双系统好像分区有区别

ptpt52, 2024-11-29:
你都没试就讲啊

qgz1, 2024-11-29:
官方系统吗

a
***@***.***

ptpt52, 2024-11-29:
> 官方系统吗 

对呀

qgz1, 2024-11-29:
a
***@***.***

 
***@***.***:~# cat /tmp/sysinfo/board_name
HCMT7981-EMMC
***@***.***:~#

ptpt52, 2024-11-29:
网口正常吗，led正常吗，mac地址情况如何，标签mac对得上吗？
测速性能如何，5G测速如何？

ptpt52, 2024-11-29:
配置 qmi 或者 mbim 网口上网

ptpt52, 2024-11-29:
目前可以配置如下，尝试一下是否可以联网
1. 删掉 网络接口的 usbwan usbwan6 两个接口

wan口修改，改成NCM协议
3. 调制解调器选择 /dev/ttyUSB3
4. Modem ifname选择 eth1
5. IP协议选择 IPv4+IPv6
6. 高级 使用网关跃点 设置 1001

ptpt52, 2024-11-30:
https://downloads.x-wrt.com/rom/Downloads/x-wrt-24.04-b202411262136-mediatek-filogic-nradio_c8-668gl-squashfs-sysupgrade.bin

更新版本，按照上述配置连5G模组

qgz1, 2024-11-30:
a
***@***.***

 

更新什么

 
https://downloads.x-wrt.com/rom/Downloads/x-wrt-24.04-b202411262136-mediatek-filogic-nradio_c8-668gl-squashfs-sysupgrade.bin

qgz1, 2024-11-30:
a
***@***.***

 
我今天又测试了一下，软件源更新报错，装插件也装不了，我自己编译一份固件刷上去也一样，软件源更新直接报错，编译上的插件识别不了架构，也更新不了

 
https://downloads.x-wrt.com/rom/Downloads/x-wrt-24.04-b202411262136-mediatek-filogic-nradio_c8-668gl-squashfs-sysupgrade.bin

qgz1, 2024-12-08:
a
***@***.***

 
c8 668编译出来固件插件openClash更新版本时识别不到处理器架构，这样就更新不了，也就不能用，有解决办法吗？

 
https://downloads.x-wrt.com/rom/Downloads/x-wrt-24.04-b202411262136-mediatek-filogic-nradio_c8-668gl-squashfs-sysupgrade.bin