x-wrt/x-wrt #435: libffmepg - doesn´t compile with support to h264.

ZacharyFoxx, 2025-01-04:
I'm trying to compile a custom version of ffmpeg to use with minidlna.

I mark libffmpeg-custom with "Y", and the options below with "Y":

    Use (L)GPL v3
    Use code with complex licensing requirements
    Include support for minidlna
    Include support for most audio decoding requirements

In Global build settings the option Compile with support for patented functionality is marked with "Y".

The compilation starts but ends with the following error:

```
libavcodec/h264_cabac.d -MT libavcodec/h264_cabac.o -c -o libavcodec/h264_cabac.o libavcodec/h264_cabac.c
{standard input}: Assembler messages:
{standard input}:138: Error: invalid operands `and $24,$7,0xC0'
{standard input}:139: Error: invalid operands `sll $24,$24,0x01'
{standard input}:140: Error: invalid operands `addu $24,$24,$17'
{standard input}:141: Error: invalid operands `addu $24,$24,$2'
{standard input}:142: Error: invalid operands `lbu $25,512($24)'
{standard input}:143: Error: operand 3 must be an immediate expression `subu $7,$7,$25'
{standard input}:144: Error: invalid operands `sll $24,$7,0x11'
{standard input}:145: Error: operand 2 must be an immediate expression `slt $4,$24,$6'
{standard input}:149: Error: operand 3 must be an immediate expression `subu $6,$6,$24'
(...)
{standard input}:4885: Error: opcode not supported on this processor: mips32r2 (mips32r2) `movz $7,$16,$6'
{standard input}:4886: Error: opcode not supported on this processor: mips32r2 (mips32r2) `movz $4,$25,$6'
make[4]: *** [ffbuild/common.mak:81: libavcodec/h264_cabac.o] Error 1
```

if I uncheck the "Compile with support for patented features" option, the compilation finishes successfully, but removes support for h264.

I know that minidlna and ffmpeg are not what we have in mind when we talk about a router, but can you give some advice on this?

ptpt52, 2025-01-05:
try `libffmpeg-full`

ZacharyFoxx, 2025-01-05:
I already tried and the mini too with same erro.

ptpt52, 2025-01-05:
build success on mediatek, also success on x86

ZacharyFoxx, 2025-01-05:
> build success on mediatek, also success on x86

Do you mind telling me what system configuration you compiled on? I'm using Manjaro.

And another thing: what was the compilation target: mt7620 (Mi Router 3)?

ptpt52, 2025-01-06:
possible can't support on mips/el
you may ask openwrt.org