x-wrt/x-wrt #442: JCG Q20 5添加led事件时出现“ReferenceError i is not defined”

cailuoli520, 2025-01-25:
root@X-WRT:~# ls /sys/class/leds/
blue:status  mt76-phy0    mt76-phy1    red:status

使用uci命令可以添加事件，例如
root@X-WRT:~# uci set system.led_blue_status=led
root@X-WRT:~# uci set system.led_blue_status.name='Blue Status'
root@X-WRT:~# uci set system.led_blue_status.sysfs='blue:status'
root@X-WRT:~# uci set system.led_blue_status.trigger='heartbeat'
root@X-WRT:~# uci commit system
root@X-WRT:~# /etc/init.d/led restart
setting up led Blue Status
luci界面里面也会出现事件，但是点击编辑还会出现“ReferenceError i is not defined”

在X-WRT v25.01_b202501010001这版本中可以正常使用，升级系统使用备份的配置文件可以正常使用

ptpt52, 2025-01-25:
感谢反馈，我有空再调查此问题

cailuoli520, 2025-02-27:
X-WRT v25.01_b202502210647版本问题已解决