x-wrt/x-wrt #444: mediatek/filogic下的11_fix_wifi_mac文件中nradio,c8-668gl重复配置

airjinkela, 2025-02-22:
```
//161	
此处	nradio,c8-668gl|\
	nradio,c8-660|\
	nokia,ea0326gmp)
		addr=$(cat /sys/class/net/eth0/address)
		[ "$PHYNBR" = "0" ] && macaddr_add $addr 1 > /sys${DEVPATH}/macaddress
		[ "$PHYNBR" = "1" ] && macaddr_add $addr 2 > /sys${DEVPATH}/macaddress
		;;
此处	nradio,c8-668gl)
		hw_mac_addr=$(mmc_get_mac_ascii bdinfo "fac_mac ")
		[ "$PHYNBR" = "0" ] && echo "$hw_mac_addr" > /sys${DEVPATH}/macaddress
		[ "$PHYNBR" = "1" ] && macaddr_setbit_la $hw_mac_addr > /sys${DEVPATH}/macaddress
		;;
```

airjinkela, 2025-02-22:
https://github.com/x-wrt/x-wrt/blob/c6fb4b763e51552563d56a42df6631f8c43ab120/target/linux/mediatek/filogic/base-files/etc/hotplug.d/ieee80211/11_fix_wifi_mac#L161-L172

ptpt52, 2025-02-22:
确实重复了，不过目前实际生效时第一组。

你有设备吗，实际运行后，mac地址是否符合预期正常？