x-wrt/x-wrt #449: 希望增加对bpi-r4 8GB RAM的支持。

aabbabaab, 2025-04-19:


ptpt52, 2025-04-19:
目前不支持8G RAM？

aabbabaab, 2025-04-20:
> 目前不支持8G RAM？
是的，我有一块8G RAM的板子，系统里只能看到3.84G，试了另一专门给8G RAM用的版本，能正常识别，显示为7.76G

ptpt52, 2025-04-21:
试试这个固件 https://downloads.x-wrt.com/rom/Downloads/x-wrt-25.04-b202504130929-mediatek-filogic-bananapi_bpi-r4-squashfs-sysupgrade.itb

aabbabaab, 2025-04-21:
> 试试这个固件 https://downloads.x-wrt.com/rom/Downloads/x-wrt-25.04-b202504130929-mediatek-filogic-bananapi_bpi-r4-squashfs-sysupgrade.itb
> 

感谢您的技术支持，新固件还是无法启用8GB内存。[image]

这是之前的固件。
[image]

这是刷写新固件之后的，仍然是4GB内存。