x-wrt/x-wrt #451: R86S使用此固件猫棒（替代光猫的直接插到R86s光口使用）无法使用

xztxy, 2025-04-30:
具体表现为 未设置猫棒接口拨号路由器正常。一旦设置拨号出现无法ping通 间歇性偶尔通一下。拔掉电信光纤，即可恢复正常。无论是官网下载的固件还是源码编译的固件都不正常。 最早21.0的固件编译的时候是可以正常使用的。 目前测试了LEDE以及IMMO的固件都是正常，不知道什么情况，麻烦大佬看下！

ptpt52, 2025-05-01:
x-wrt 支持这个型号？

xztxy, 2025-05-01:
> x-wrt 支持这个型号？

支持的。这个就是一个x86 小主机 之前都用的没问题，由于很长时间没更新了，更新后发现问题。之前的应该是21.0的版本。

ptpt52, 2025-05-01:
日志
logread
dmesg

xztxy, 2025-05-01:
> 日志
> logread
> dmesg

好的，晚点等家里人不用网络的时候我把日志弄一下，麻烦您了！

xztxy, 2025-05-01:
> 日志 logread dmesg

[日志.txt](https://github.com/user-attachments/files/20001341/default.txt)
2个命令的日志都在文本附件中， 刷了固件后 配置好eth3.318 设置了wan pppoe 拨号 然后路由器失去连接，我再去拔掉电信光纤，路由器恢复连接，然后使用 logread 和 dmesg 获取的。麻烦大佬看下什么情况，谢谢！

xztxy, 2025-05-02:
> 提供x-wrt官方固件的日志，提供出问题时候的日志

这个日志是官方固件的日志。，之前的日志也是官方源码，只是自定义了一些软件。

[log.txt](https://github.com/user-attachments/files/20010949/log.txt)

ptpt52, 2025-05-04:
看不出，可能是网卡驱动问题

ptpt52, 2025-05-04:
openwrt 官方最新的固件可以正常吗？

xztxy, 2025-05-04:
> openwrt 官方最新的固件可以正常吗？

没试过，不过lede和immo的正常。

ptpt52, 2025-05-04:
用 openwrt 官方最新固件对比一下看看